uavpal/disco-docs #7: Streaming your video

djoris, 2018-08-14:
You can grab the video from your Skycontroller (or Disco?) using:
http://SC2:7711/video (this is a referrer file, linking to 192.168.53.1:55004  (H264 stream) )

command to stream to YouTube:
ffmpeg -f lavfi -i anullsrc -rtsp_transport http -i rtsp://DISCO-OR-ZEROTIER-ADDRESS:55004 -tune zerolatency -vcodec libx264 -t 12:00:00 -pix_fmt + -c:v copy -c:a aac -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/STREAM-NAME

See: https://www.georgetimmermans.com/blog/streaming-rtsp-to-youtube-live

djoris, 2018-08-18:
@uavpal 's:
ip_sc2="a.b.c.d" # either SC2 zerotier IP or USB Ethernet IP

# required to trigger video streaming to my IP
curl http://${ip_sc2}:7711/video 2>/dev/null | sed "s/addr:192.168.53.1/${ip_sc2}/g" >/tmp/video.sdp

ffmpeg -f lavfi -i anullsrc -protocol_whitelist file,rtp,udp -i /tmp/video.sdp -vcodec h264 -t 12:00:00 -pix_fmt yuv420p -c:a aac -preset ultrafast -b:v 1000k -bufsize 512k -vf scale=854:480 -g 60 -f flv rtmp://a.rtmp.youtube.com/live2/abcd-abcd-abcd-abcd

djoris, 2018-08-22:
alternative is to stream from your phone/tablet. Easier, but doubles the data usage on your mobile (and heavier on the processor as well). [Louis Mcnair](https://www.youtube.com/channel/UCHOKayO20xX-FK867C5YJHw) uses https://obsproject.com/ to mix streams.

djoris, 2018-08-26:
can you stream from the Disco instead of SC? That would save on latency and bandwith https://developer.parrot.com/blog/2016/play-bebop-stream-on-vlc/
how to trigger the stream?
Experimenting with SC2 stream in VLC for windows shows distortion, probably not enough buffer. Also, only one stream at a time seems to work?

ClassAxion, 2021-06-09:
Video stream uses RTP H264. You have to trigger the video stream then disco starts `UDP` streaming from `DISCO_IP:5004` to `YOUR_IP:55004`. 

To trigger the video you can use this example: https://github.com/hybridgroup/node-bebop/blob/master/examples/video.js#L13 but you have to modify this parametres: https://github.com/hybridgroup/node-bebop/blob/master/lib/bebop.js#L266 adding this:

```
      "arstream2_client_stream_port": 55004, 
      "arstream2_client_control_port": 55005, 
      "arstream2_supported_metadata_version": 1, 
```

So you sending your controller info to discovery port `444444` with stream ports and then you trigger the video using `VideoEnable` command. After this operation disco starts pushing data over UDP. Remember to turn off your local firewall :) 

The next step is to check what the `:7711/video` endpoint on SC2 is doing. I'll check it with wireshark. It is possible that it triggers the video in a different way than described above.