uavpal/disco-docs #9: understand temperature measurements

djoris, 2018-08-20:
the Altimeter Module MS5607 and the MPU-6050 accelerometer respectively, can show their temperature values:
when smothered:
```
/ # diagnostic -v|grep -i temp
posix init start build on : Jun 20 2018 15:39:44
Use ctrl+\ (SIGQUIT) to end the application
        | C6 : TEMPSENS        | 0x6BFA | 27642 |     28165    |
        temperature: 66.412354 degree Celsius
Segmentation fault
```
and when properly ventilated on the bench:
```
        | C6 : TEMPSENS        | 0x6BFA | 27642 |     28165    |
        temperature: 58.130001 degree Celsius
```

djoris, 2018-08-20:
Request: test this in the air with Disco4g

djoris, 2018-08-20:
another sensor: 
`BLDC_Test_Bench -O | grep -i temp`

djoris, 2018-08-21:
https://github.com/uavpal/disco-docs/wiki/CHUCK

djoris, 2018-08-22:
alternatively, the black box is said to give temperature readings.
/etc/debug.conf: BLACKBOX=1 (or /usr/bin/DragonDebug.sh)
each flight will be stored in /data/ftp/internal_000/blackbox , then convert to csv:
https://github.com/AutonomyLab/bebop_hacking

s0ftice, 2019-05-13:
BLDC_Test_Bench also returns
`*** Warning: accessing to settings is forbidden when motors are spinning ! ***`

Not so keen to try in the air ... :)