uavpal/disco-docs #10: Start flight plan without paying $20 in FreeFlightPro3

djoris, 2018-08-22:
You can create flight plans with Groundcontrol for Arduflight (?) and upload them (FTP port 61?)

You have to trigger start/pause/stop for flight plans without FFP
* use the example python code? https://github.com/N-Bz/bybop
* send this call? https://developer.parrot.com/docs/reference/all/#start-a-flightplan

ClassAxion, 2021-06-06:
In theory, you can just rebuild the android apk and throw away the part responsible for the payment. However, in my opinion, it already involves piracy.

ClassAxion, 2021-06-09:
```
public void startFlightPlanIfPossible() {
    if (this.mFlightPlanRegistrationManager.isPurchased() || this.mFlightPlanRegistrationManager.isPromotionActivated()) {
        startPiloting(3);
        return;
    }
    ((AppCompatActivity) this.mContext).startActivityForResult(new Intent(this.mContext, FlightPlanFreeTrialActivity.class), 2);
}
```

Just remove the condition :) 

PS. @djoris I'm curious what makes you laugh in my comment ;)

kikislater, 2021-06-09:
Could you please add source to apk app ?

ClassAxion, 2021-06-09:
This code is located at `com.parrot.freeflight.home.HomeUIController.java`.

kikislater, 2021-06-09:
Ok thank you but where do you find source code for Freeflight pro ? Is there a repository ?

ClassAxion, 2021-06-09:
Of course not, you have to do it yourself.

kikislater, 2021-06-09:
So you mean file in smartphone, this is not rebuilding apk ....

ClassAxion, 2021-06-09:
No, it's rebuilding - I don't edit any file on smartphone directly. I create a new apk that I install on my smartphone (of course I have to uninstall previous version first). 

What's stopping you from decompiling it yourself, change what you need, and compile it again? 

However, for apk there are easier methods. You can use for example `APK Easy Tool` that can decompile APK to Smali and compile again to APK. In this case, you don't have to worry about dependencies and that's what I'm using.

kikislater, 2021-06-11:
Interesting, didn't know this procedure, 
Thank you for sharing