uavpal/disco-docs #12: disco I2C ESC limited lets find pwm!

stephonunlocker, 2018-09-08:
Hello i am trying to find a hardware or code solution for a project i am doing with the parrot disco .
i want to use a secondary regular ESC on the parrot c.h.u.c.k  so i can use bigger motors that draw more AMPS .
i know that the reason they used the i2C ESC on the disco is so they can reverse the motor on landing i am in the process of using my oscilloscope on the c.h.u.c.k to see if  can find a usable pwm motor control signal to tap into if present or if not is there a code that i can use in the original dragon firmware to output motor pwm signal on one of the free pins close to the servo controls .
the goal is to use the disco c.h.u.c.k in bigger air frames with bigger motors/ batteries and utilize the free flight pro and flight plan . i know its possible to use esc in other modes but i want to maintain the use free flight pro and flight plan i dont want to use rc  .
 
https://community.parrot.com/t5/Disco-Knowledge-Base/Configuring-APM-for-Parrot-C-H-U-C-K/ta-p/161782
[image]

djoris, 2018-09-10:
Doing a hardware mod would be awesome! Haven't seen anyone try that yet. As far as controlling the other pwms, I hope we can get there (#8) then in theory, doing a readout of channel 3 and copy that value to another channel would suffice (but you'd still have a bit of energy loss from that unused ESC?)

BTW I thought the stock ESC isn't going into reverse on landing; the props flop when the direction is reversed so it would have little use.

Please keep us updated! :)

stephonunlocker, 2018-09-10:
i think it does reverse on landing to slow it down from glide its one of the main feature of i2c.  energy loss from that unused ESC might be minimal i not sure what behavior i will get but i might have to put a fake load not to get a motor error.

AndKe, 2018-11-02:
This would be easier if we had the wifi-controls working with ArduPlane, and SAFERTL would be a huge bonus... So maybe its more worth your time to research the ArduPlane , than hacking the proprietary autopilot code..

hamelsveld, 2019-01-28:
And did it work? I also would like to add an external ESC of 60A to the chuck! but have no idea how to do so...

masterpii, 2019-03-29:
Hey all. I also tried to mount a wemotec/Hacker impeller on the disco. Ignoring the flight physic downsides the motor draws too many amps. 37A as far as the data sheet tells. Using two standard batteries that’s not the blocker to supply that. However the current limiter stops the run-up. So I actually have the same idea. If one could use one of the servo pwm signals to control a brushless Controller using it as an power amplifier that woluld be awesome. I also want to maintain the current software environment. Does anybody know how to access the esc Controller to modify its parameters. I guess the amp limit is set by software