uavpal/disco-docs #14: Cruising current rating seems and Li-ion batteries

mikerobertsn, 2019-04-22:
Hi, just looking at the crusing current rating you gave here https://github.com/uavpal/disco-docs/wiki/batteries, it seems to be more like half that, 45 minute flight time -2.7Ah, 2.7/0.75, average current draw would be a maximum of 3.7A.

Also I saw you had the titan battery there, just wondering if anyone has any success in using a 3S1P configuration, any heat issues, going by the specs it seems unlikely to create problems.

djoris, 2019-04-22:
Thanks for the input, I've updated the wiki! The calculation seems obvious, no idea where the 6A came from... perhaps someone estimated 6A is used when climbing?

I think I remember someone using a single pack of Titan, but I'll ask on [Slack ](https://uavpal.slack.com/messages/CCKRS8VQ8/) because I'm not sure.

kikislater, 2019-05-27:
Computation is not good : fly with disco then charge Lipo with a proper charge which read volt and amps. Then you will know exectly how many mAh you consumed.
Max Amps could not be calculated like that, you need to test it on the ground with power meter. It was done on rcgroups thread and it was around 10-12 if I remember well

mikerobertsn, 2019-05-27:
No one was suggesting that you calculate max amps like that.
Not sure how you do it on the ground since there isn’t direct control over the throttle, pressing takeoff and holding it I’d imagine it would only give you an approximation, as you would need to be flying to be able to measure the max current that it while flying.

kikislater, 2019-05-27:
Boot on ardupilot firmware and you will able to arm your aircraft then mesure max Amps with external power meter ...

gus7119, 2019-05-31:
Couldntyou just bypass CHUCK use external RX and get amps via taranis telemetry modual? 
Thought that had a been settled anyway with 3.6 being cruise and 15amp peak on take-off, thought that was the agreed amp usage.
Not trying be a dick just assumed that was what we where working from.
But anyone with a taranis should be able to settle this in a couple of minutes.

kikislater, 2019-06-03:
No ! on board amp report is not as accurate as an amperemeter could do it ... 
Check FPV guys, it's common to measure it like that

Here is a proper test : https://www.youtube.com/watch?v=2i0zmSclT8g