uavpal/disco-docs #16: LiHV Battery - Last motor error battery voltage out of bounds

NFK007, 2019-06-21:
Hi, I am using a Turnigy 3S2P 5200mah LiHV battery - and balance charge to full capacity 4,35v per cell = 13,05v.

I had no error during the first flight, but now after a full charge the Parrot Disco complains with error: Last motor error - battery voltage out of bounds.

I have tried modifying the parameters.conf file, but to no effect.

Anyone have an idea on how to fix?

s0ftice, 2019-06-22:
Maybe it was not charged fully the first time you tried. Or you charger was set to LiPo instead of LiHV. If you charge it fully as LiHV, the Disco (ESC/Motor) will not accept it. Interestingly, the Bebop2 is ok with it.
There is no software modification you can do, it’s really the hardware refusing... You can try to charge it to “LiPo” max voltage and it will work. Or just get a LiPo instead of LiHV...

djoris, 2020-07-13:
I finally made explicit in the documentation what s0ftice said above

closed 2020-07-13 by djoris