uavpal/disco-docs #18: Angle settings with heavy batteries

CurlCie, 2021-04-03:
Hello

I added a 7000 mA lion battery (heavier than the original battery)

I have some troubles when the disco takes off : only 5 secondes after take off, the disco pitching too much and crashes

Do you know how to fix it ?

The gravity point is the same compared with the original battery

Should I modify the : TAKE_BEST_ANGLE_GOAL_PITCH from 30 to 20 ?

Is there a time limit to reach the good altitude ? 

Thank you very much !

CurlCie, 2021-04-04:
Rolling and not pitching... poor english !

calypsoopenmail, 2021-04-05:
What battery reference is it?

CurlCie, 2021-04-05:
Batterie 11.1v 7000mAh Li-ion Samsung 3S2P
[https://www.ebay.fr/itm/RC-Batterie-11-1v-7000mAh-Li-ion-Samsung-3S2P-prise-balance-XT60-Parrot-Disco/124208650442]

it is oscillating, each oscillation is bigger than the last one

First, I thought the chuck was broken, but I installed a new chuck and the problem is the same.

even if I try to compensate with the Skycontroller, results are worse ! 

Thank you for your help !

calypsoopenmail, 2021-04-05:
Same for me. I have 3S2P Kluosi. The servo has been modified, but no problem with stock lipo battery. Every screw is fine, everything is locked, nothi,ng move inside. I was suspecting the current was not enough with li-ion. But many people seems to fly well with similar configuration. Maybe calibration?

Have you changed the servos?

thewillhuang, 2022-01-29:
NAV_DEFAULT_RADIUS                                          = 50.0                                                                             
NAV_DEFAULT_AIRSPEED                                        = 16.0  
NAV_DEFAULT_MAX_TURNRATE = 20

zagor, 2022-01-31:
I added a section to the battery page recommending changing the above parameters: https://github.com/uavpal/disco-docs/wiki/batteries#configuration-parameters

closed 2022-01-31 by zagor